DUNE/dune-tms #12: Ability to overlay events on top of each other

clarenc3, 2022-02-23:
Right now we have single events which are pretty alright to reconstruct. In reality, we'll get multiple events per spill. The simplest way to accomplish this in the code is to either combine edep-sim events before running, or combining them in the edep-sim code.

jdkio, 2022-09-14:
There's a way to do this in the ND_Production scripts. I'm not sure how they're handled in dune-tms, other than to say that it runs. I assume it gives all events a time of 0 and considers it a single event. So the next step would be give them a random time according to some spill structure, and the split them into individual events using a time slicer.

jdkio, 2022-09-14:
This should also include a model of deadtime, which can be a large effect in terms of tracking efficiency.

jdkio, 2023-03-09:
This works now. The edep sim step gives a random number of interactions and at a random time according to a spill struction. I've made a new issue out of deadtime.

closed 2023-03-09 by jdkio