DUNE/dune-tms #19: Improve event display

clarenc3, 2022-09-12:
Would be really nice to have a proper event display (TEve based?), which shows the reco and true hits and tracks in a configurable manner. This doesn't seem *too* difficult, but needs the time.

jdkio, 2023-03-09:
The timing branch has scripts/TimeSlicer/draw_spill.py which will draw TMS events for specific spills. It doesn't draw the corresponding LAr detector spill. It's intended to study the effectiveness of the time slicer. Also it only prints pictures. But it can be used as a jumping off point for another event display.

AsaNehm, 2024-02-13:
Developing an event display that shows tracks projected from 3D into 2D plots (matplotlib/python)

AsaNehm, 2024-07-09:
Projection from 3D into 2D plots finished

AsaNehm, 2024-09-09:
@LiamOS @jdkio I think we can close this as there is a new version of the event display by now

LiamOS, 2024-09-09:
I'm inclined to leave it open, your event display is great, but what we (+ any and all friends of TMS) really want is a big 3D thing that shows the relevant detector geometry together with all the tracks in a GUI that allows us to step through events, etc.

Nominally this work is on me, and since yours works more than well enough to debug most issues I've been slacking.
I've self assigned this to increase the shame I should feel for every day this is not done.