DUNE/dune-tms #20: Save truth information separately from reco

jdkio, 2022-09-14:
Right now there is no distiction between reconstructed hit position/energy/time and true position/energy/time. 

In the future simulation, the optical model would use true energy deposited to estimate the reconstructed energy. For backwards compatibility, the current hit variables should be considered reconstructed variables, and truth information should be added as new variables.

In future data taking, the data wouldn't have the truth information. So the truth branches shouldn't be saved in that case.

jdkio, 2022-09-14:
Some of this is already there. True hit saves E and position, but not time. With an optical model, it should also save true PE as well, which would be used for time skew.

AsaNehm, 2024-09-09:
@jdkio What is the status on this atm?

closed 2025-02-04 by AsaNehm