DUNE/dune-tms #23: Memory leak when running with overlay

jdkio, 2022-11-13:
There's a memory leak somewhere that prevents the TMS reco from running with overlay. This was not an issue with the last sample run with overlay so it must have been introduced recently. That was run circa July but it's not clear if I was running with an older version of the reco.

For example,
ConvertToTMSTree.exe /pnfs/dune/persistent/users/kleykamp/example_edep_file_with_overlay.root
will crash because it runs out of memory.

clarenc3, 2022-11-14:
During the meeting today we said that it likely is simply due to the event size; the time slicing should really happen before the reco stage. Will look into this

jdkio, 2022-11-30:
Turns out it's not a memory leak nor due to event size. TMS_Event loops over all primary vertices and then within that loop gets *all* the hits, not just the subset associated with each vertex. So you end up with ~190x as many hits per event which runs out of memory before finishing the loop. Moved the code that loops through hits outside the primary vertices loop to fix.

jdkio, 2023-03-09:
This is fixed in the timing branch, which should be merged soon. Closing this isssue.

closed 2023-03-09 by jdkio