DUNE/dune-tms #54: Make sure dumpSSRITree.cpp info makes it into truth_info and then cafs

jdkio, 2024-01-12:
[dumpSSRITree.cpp](https://github.com/DUNE/dune-tms/blob/main/scripts/FlatTree/dumpSSRITree.cpp) is a truth-level script that takes edep sim files and converts them into a tree. For example, it tells you the exiting momentum from the LAr, and the momentum starting into the TMS. This is useful for certain types of studies. 

I think this is the script that Chris M used to give Tom L a file for his studies.

Ideally we want all this info in the truth_info in TMS. And then in the cafs. That way we can do these studies using official files and official cafs. So the dune-tms tree writer should save this kind of info, modeled on dumpSSRITree.

LiamOS, 2024-01-12:
Assigned myself. (volunteers absolutely welcome!)

Useful truth info that is correctly associated with reco objects will be absolutely necessary in the medium term.

jdkio, 2024-01-18:
I have some ideas on this and was going to attempt to add the following

per particle:
pdg
true start position (4 vec with time)
true end position
true start momentum (4 vec with energy)
final momentum?
index to true parent (-1 if no parent)
number of children?
reaction string? Info about what type of interaction this particle died with, or about interactions
number of true TMS hits from this particle
simple flags if it hit the LAr/TMS
total visible energy inside the TMS from this particle
true pos/momentum of first true hit (lowest z), and last true hit (highest z)
true pos/momentum on entering LAr, TMS (that's 16 numbers, -999 if wasn't in LAr)
true pos/momentum on leaving LAr, TMS (that's 16 numbers)
true pos/momentum on transitioning from TMS thin to TMS thick? And other special positions. Like maybe within the first two modules
In theory things can leave TMS and then bend back. Are we worried about that?

per reco track:
true visible E deposited per true particle (list)
"other energy" (for dark noise or data overlay) 
index to true particle (list)
Instead of saving the full list, we might only save an index that points to the primary particle.

Info per true particle assuming highest true energy particle is the primary particle:
primary particle true visible energy 
total true visible energy from all particles
energy ratio that between primary particle true visible energy / total true visible energy (to cut "pure" particles vs tracks that are composits of multiple particles).
ratio of primary true visible energy / true visible energy from that particle (which lets you cut on reco tracks that only have part of a true particle)
true momentum at start z - lets you check the p reco for just the segment that was reconstructed
true momentum at reco end z - lets you see exactly how much energy was missed

closed 2025-02-04 by AsaNehm