DUNE/dune-tms #78: Where is TMS in y coordinates?

AsaNehm, 2024-03-27:
With the starting studies of different geometry options regarding layer orientation, the y coordinate of TMS becomes more important to run small batches of muon guns to test out the geometry quickly.
This information is also important for the track matching in 3D as this uses the anchor points of the scintillator bars in y to calculate the y coordinate of a hit.
Unfortunately the available information on where TMS is in y is very confusing.

- PDR (Doc-DB 25546) states that "TMS is placed 8.2m downstream from the ND-LAr (center-to-center) and centered 2.51m below its origin in y." -> Where is ND-LAr's origin?
- geom_v_1.0.3 
[image]
-> Scintillator bars are supposed to be 3.2m in length. This is apparently not the case in this implementation
- JSROOT geometry browser
[image]
-> Something completely different and this are also not global coordinates as shown here
[image]
- Tom LeCompte: "Intent is for the center of the thin-thick plate transition (plane 40) be 6 degrees below the center of the ND-LAr fiducial." -> Where is ND-LAr?

ast0815, 2024-03-27:
I think really, the only reliable thing to do is to figure out how to ask the geometry volume elements where they are in the global coordinate system. There must be some way to do that. While doing that, we might also want to think about how the software handles TMS being at different x positions. Should our reconstruction care? Should our output be tracks in global coordinates or in local TMS coordinates? If it is local TMS coordinates, we could also define our y_0 how ever we want.

jdkio, 2024-04-02:
Looking at the geom v3,
`/pnfs/dune/persistent/physicsgroups/dunendsim/geometries/TDR_Production_geometry_v_1.0.3/nd_hall_with_lar_tms_sand_TDR_Production_geometry_v_1.0.3.gdml`
The scintillator "box" is defined here: `<box name="scinboxTMS" lunit="cm" x="3.542" y="309.6" z="1.0"/>`
So a scintillator bar is 309.6cm long, by 1cm deep, by 3.542cm wide. We can change it to match the 3.2m length but that would require that we redo the bar overlap study. I don't remember who did that. The y vs z track start plot includes starting positions from steel so it overestimates the size of the scinillator bars

Currently we use local TMS coordinates. This avoids issues like when the TMS is moved off-axis. The track matching alg can probably be responsible for putting the tracks into global coordinates

`TMS_Bar.cpp` has code that deals with the size and position of the scintillator strips. We can put code similar to that into `TMS_Geom.h` that returns the y_0 position, and useful information like the actual length of the scintillator bars

jdkio, 2024-04-03:
For the unrotated geometry, the center of a strip is at y=-1352.23. This is based on printing the position
`geom->GetCurrentMatrix()->LocalToMaster(local, position)`
where local is (0, 0, 0) when geom's node is a scintillator bar. So this should be translating the local origin into the global coordinate system. Comparing to true geant4 hit position below gives me some confidence that it's doing the right thing. Here the hit can be anywhere in the bar, but then the second tuple is rounded to the nearest bar. You can see that the x and z positions agree within 35mm and 10mm respectively.
```
Hit position: (-1828.90, 70.76, 13591.60), bar position: (-1839.13, -1352.23, 13593.00)
Hit position: (-635.53, -2881.24, 12466.35), bar position: (-624.85, -1352.23, 12468.00)
Hit position: (-1968.80, -1392.45, 15832.90), bar position: (-1980.81, -1352.23, 15833.00)
Hit position: (1129.80, -2806.06, 17431.47), bar position: (1120.73, -1352.23, 17433.00)
Hit position: (-1625.05, 56.06, 16392.30), bar position: (-1616.61, -1352.23, 16393.00)
Hit position: (2054.15, -1716.06, 11919.18), bar position: (2051.65, -1352.23, 11918.00)
Hit position: (2630.83, -307.29, 17429.83), bar position: (2618.37, -1352.23, 17433.00)
Hit position: (-2183.74, -821.74, 12793.05), bar position: (-2193.33, -1352.23, 12798.00)
Hit position: (-1324.31, -276.72, 13834.69), bar position: (-1333.25, -1352.23, 13833.00)
```

When using geometry v3, we get changing y positions. That's because the bars are rotated as a group. So the center moves downward at 3 degrees. This also affects the x position of the bar so it's no longer within 35mm.
```
Hit position: (1022.39, -2679.60, 12911.05), bar position: (1084.99, -1364.28, 12908.00)
Hit position: (1046.17, -2619.75, 13123.78), bar position: (1120.37, -1366.13, 13128.00)
Hit position: (861.31, -2815.16, 12966.62), bar position: (802.02, -1355.01, 12963.00)
Hit position: (961.61, -2903.63, 13017.36), bar position: (1049.62, -1362.43, 13018.00)
Hit position: (953.24, -2875.47, 12967.26), bar position: (872.77, -1351.30, 12963.00)
Hit position: (952.21, -2686.23, 12690.52), bar position: (1014.25, -1360.57, 12688.00)
```
But they're all roughly -1352.23 and I bet the average location is exactly that since the they pivot from their centers. But I haven't checked

LiamOS, 2024-04-08:
It might be good to save the offsets from GDML origin to TMS origin (we can discuss what point in the TMS to choose) when this is run, so that the files include a way to transform from 'global' hall coordinates to 'local' TMS ones.

AsaNehm, 2024-05-13:
Summary
x: from -3520 to 3520 [mm]
z: from 11362 to 18294 [mm] with beginning of thick steel at 13500 [mm]
y: steel from -3864 to 1159 [mm], subtract from both ends ~915 [mm] -> scintillator from -2949 to 244 [mm]