DUNE/dune-tms #94: Reco hit pos doesn't make sense for UV planes

jdkio, 2024-04-18:
Reco hit pos is storing x,y,z of hits, but x and y are u and v in some cases. So we need a better way to store things. 

https://github.com/DUNE/dune-tms/blob/main/src/TMS_TreeWriter.h#L233

One option is to add another branch that tells you whether the hit was a X or U plane. Alternatively, we can explicitly save u and v hits in difference branches. So that x/y are -9999999 when u/v aren't and vice versa.

AsaNehm, 2024-04-22:
I tried to mitigate this problem already by storing the reconstructed x and y for u and v hit in reco hit pos (see Line 965ff in  TMS_TreeWriter.cpp).
Do you think this helps for this?

AsaNehm, 2024-09-09:
@jdkio Is this still an issue for you?