DUNE/dune-tms #107: Reco Track Direction is Not Correct

jdkio, 2024-05-10:
The reco track direction variable code is [here](https://github.com/DUNE/dune-tms/blob/main/src/TMS_Reco.cpp#L1261-L1263). This is not how to calculate direction. This is giving the overall  average direction of the track. But it includes all the scattering and curvature from magnetic field. We want the direction roughly at the start of the muon track. That's the difference between the blue line and the dotted line below.
[image]

Each node of a track has a position and direction. So the direction at this node would be measured by doing dx/dz between two consecutive nodes at the start of the track. It should match as closely as possible with true px/pz of the muon at the start of its track.

One caveat is that in some cases you don't want to look at the very first two nodes. Sometimes if there's a lot of activity, the reco track isn't measuring the direction of the muon, but some average direction. In those cases, it makes sense to measure the "starting" direction of the muon a few nodes downstream. For Minerva (which is pure scintillator and has no magnetic field), we measured the direction 20 planes downstream to avoid hadronic activity. For TMS, this distance would be shorter to avoid scattering and b-field curvature.

AsaNehm, 2024-05-13:
When I implemented this initially, I wasn't aware of how this would get used by people and therefore implemented this in a way that made sense to me (mostly for plotting purposes). But this absolutely can be changed and should be if it makes more sense in a different way.
@jdkio if you can provide me with a full explanation on how this should be done correctly (not with code, just the method), I can implement this. Otherwise I would prefer, if you would implement this as you seem to know what is required more than I do

jdkio, 2024-05-13:
You know more about the structure of the reco track class than I do. The way I imagine it works is that you have a list of hits associated with a track ordered from first to last hit of a track. In that case, the direction dx/dz would be 
```
dx = hits[0].x - hits[1].x // replaces hits[0].x - hits[len(hits) - 1].x
dz = hits[0].z - hits[1].z
```

AsaNehm, 2024-05-14:
Created a new branch for this and implemented it in this way
To allow for easy future tuning of the exact distance from the start to calculate the direction from, I added a parameter in the config file called "DirectionDistance". This gives the number of planes that the direction is calculated for away from the start.
@jdkio Is the direction actually supposed to be in this way hit[0].z-hit[10].z with hit[0].z < hit[10].z? Otherwise I'll turn it around