DUNE/dune-tms #113: Weird track length behavior NERSC files

AsaNehm, 2024-05-15:
The performance plots for the track momentum/length reconstruction show a weird behavior
[image]

AsaNehm, 2024-05-15:
(full output file at /exp/dune/app/users/pnehm/output_UV_NERSC.txt)

AsaNehm, 2024-05-15:
Checking for the truth for each iteration first if both hits are inside TMS (TMS_Geom::IsInsideTMS) seems to help
[image]

jdkio, 2024-05-15:
One thing I found is that the geometry gets corrupted somehow if it tries to measure a position far from the TMS. When that happens, all future calls are broken. This seems to be what's happening here. I fixed it by limiting the z in the world

The reason this happens with the nersc files specifically is because they have rock muons. The rock muons are more likely to reach this boundary where the corruption occurs

AsaNehm, 2024-05-16:
Still a smaller 0 bin for the truth remains...

AsaNehm, 2024-05-17:
For the full first NERSC samples 610 truth events have a density-weighted track length of exactly 0.
There are definitely also a non-negligible number of events where the reconstruction gets a considerable density-weighted track length while the truth stays at 0.
This seems to me like there is something bad going on with the truth

AsaNehm, 2024-09-09:
I think this issue has been solved finally

closed 2024-09-09 by AsaNehm

AsaNehm, 2024-09-09:
This was at least no longer observed in recent looks at the tracklength. There were changes to the truth information and changes to the reconstruction. Checked out with merge 135