DUNE/dune-tms #114: What is going on with the track length?

AsaNehm, 2024-05-16:
Looking at Magnus' hybrid files (particle gun, muons with ~1GeV, all starting at roughly the same point in TMS) reveals in comparing truth and reconstruction for density-weighted track length something unexpected (the disagreement for pure stereo gets worse here by a lot)
[image]
The truth having this spread is rather unexpected and that the reconstruction is mostly so far off is also worth noting.
Looking at the track length in mm (just from the distance) does not explain the difference
[image]
The true track length is calculated in TMS_Event.cpp GetMuonTrueTrackLength(), the reconstructed track length in TMS_Reco.cpp CalculateTrackLength3D(). Both functions refer back to TMS_Geom.h GetTrackLength() and are methodically identical.
Plotting the track length in mm against the density-weighted track length looks for the reconstruction like this
[image]
and for the truth like this
[image]
What is causing this?

jdkio, 2024-05-21:
Where are Magnus's files and which tag of the code was used? That would help me with debugging
Sometimes Muon_TrueTrackLength isn't filled correctly. Like I know that with the time slicer it didn't fill correctly. In the future, it should be deprecated and instead we'd use the true path length associated with the reco track.

Here are the plots I get on my local branch using the true path length associated with the reco track for true muons fully contained in the TMS. It doesn't have recent reco changes, which I think explains why reco doesn't look right. 
[image]
[image]
[image]

AsaNehm, 2024-05-21:
Magnus hasn't uploaded his files to anything at the moment unfortunately. I can try to talk to him about it and set this up.
For the tag of the software I'll also ask him as I'm not sure of that either.

The time slicer being the 'problem' for this makes sense and should explain it indeed.

The true path length associated with the reco track, you mean the one from TMS_Track?

jdkio, 2024-05-21:
In my branch (which I hope to finalize soon), I added RecoTrackPrimaryParticleTrueTrackLength, which is the track length from where the reco track starts to the true end of the particle track

AsaNehm, 2024-05-21:
That would be really useful. Looking forward to that :+1:

magnushandley, 2024-05-21:
Heya - the files I give Asa are just the edep-sim outputs, so no reco has happened at that point. I then run the reco separately to produce my own plots, but Asa will know which version they used to produce the plots above. I've dumped a few of the muon gun samples here: /exp/dune/data/users/mhandley/energy_angle_variations. Any of the ones with hybrid in the name include orthogonal planes. The geometries used to create them are in /exp/dune/data/users/mhandley/full_hall_nosand_geometries

AsaNehm, 2024-05-21:
The time slicer comes only into play during the reco and not in the edep-sim output, correct?
I run the files on the newest version of the software that I have available which is usually either the main branch or whatever thing I added on my own branch from the main branch. So this should already be with the time slicer problems fixed

AsaNehm, 2024-09-09:
I think this also has been fixed finally