DUNE/dune-tms #120: Hits included in tracks that shouldn't be there

LiamOS, 2024-06-13:
In file`/pnfs/dune/persistent/users/kleykamp/nd_production/2024-05-13_lar_only_rhc/edep/RHC/00m/00/antineutrino.0_1715654453.edep.root` for spill 10, the reconstructed track is as shown:
[image]
This happens in a number of other events in other files, and also happens in the NERSC sample files.

AsaNehm, 2024-07-01:
Starting to work on this

AsaNehm, 2024-07-01:
It seems that there is an effect that 'randomly' assigns the x position of a hit to 0 at the end. This is an effect and not real as this neither appears in the truth hits nor in the hits for reconstruction before the reco starts. After the 3D matching this effect is there. I'll trace this through and fix it

AsaNehm, 2024-07-02:
I think I managed to solve the issue. See branch 120-...

AsaNehm, 2024-07-04:
Forgot to fix the 'random' hits in the middle of tracks. Now taken care of by handling two hits per plane in the matching step correctly

AsaNehm, 2024-07-17:
Will be merged into the main branch with PR 122

closed 2024-07-17 by AsaNehm