DUNE/dune-tms #138: True Muon KE

mdolce8, 2024-08-30:
There is something not quite right with the true muon energy as it enters TMS. 

I am seeing from running 
`python draw_spill_3D_projections.py --outdir /exp/dune/data/users/mdolce/dune-tms/spills/truth  -f /exp/dune/data/users/kleykamp/tmsreco_combined_files/2024-04-19_bfield_1p0T.tmsreco.root --only_true_tms_muons` some unusual (and big) muon KEs from the truth as they enter TMS. I have pasted some of the output I was seeing below. I would say this happens around 10-20% of the time. So there is something not quite right with the true muon energy. It is calculated by using the `MomentumTMSStart` branch from the truth. 

I ran this code from the branch: `mdolce_drawspills_truehits`. I added a few lines of code to Asa's script that loops through the `Truth_info->PDG` for each event and retrieves the muon true energy. And that was how I found this issue.

Here are a few examples of the issue: 
```
Event: 0, Spill 6913, Muon KE at birth (LAr): 26.086267578125, Muon KE entering TMS: 173204.97505692, GeV.
Event: 0, Spill 8722, Muon KE at birth (LAr): 1.402595703125, Muon KE entering TMS: 173204.97505692, GeV.
Event: 0, Spill 8875, Muon KE at birth (LAr): 3.2294921875, Muon KE entering TMS: 173204.97505692, GeV.
Event: 0, Spill 9899, Muon KE at birth (LAr): 2.68086328125, Muon KE entering TMS: 173204.97505692, GeV.
```

jdkio, 2024-08-30:
This may be fixed by this PR
https://github.com/DUNE/dune-tms/pull/134#issue-2490277765
We've fixed many issues since we made these files. We should remake the files with the most up to date code

jdkio, 2024-08-30:
Momentum inside TMS or whatever it's called works better. It's not the momentum at the z position at the start of the TMS. Instead, it's the momentum when it crosses into the TMS fiducial, regardless of angle. For lar starting events, this is the same usually

mdolce8, 2024-09-03:
Is it possible to remake the files with the up to date code? Maybe after the collaboration meeting?

Also to your second comment, I think that is what I used: `truth.MomentumTMSStart` to get the momentum, and calculate the KE from there. So I think this is right.

jdkio, 2024-09-13:
Yeah, this is long overdue. I'm working on a validation suite and I'm still uncovering issues. But maybe we can't let perfect be the enemy of good

closed 2025-02-04 by mdolce8