DUNE/dune-tms #147: Remove bar size hardcoding

jdkio, 2024-09-05:
[If we run](https://dunescience.slack.com/archives/C041ZU59LLV/p1724789614407889) with any other size bar (like when we adjust the geometry), we get

```
width of scinBoxlvTMS_PV_13 not as expected!
xwidth: 30.2
ywidth: 3096
zwidth: 10
terminate called without an active exception
```
That comes from this hardcoded check here:
https://github.com/DUNE/dune-tms/blob/5ab577fd4610fd1c5bb66abd4c7c2371cdde95c5/src/TMS_Bar.cpp#L78
That needs to be removed

LiamOS, 2024-09-09:
Am adding myself and @AsaNehm on this, the Kalman filter will need a small modification to work with more general bar sizes. I imagine there are some numbers/indices/values in the reco code that assume bar geometry and/or angle that should be generalised.

Might be needed to allow TMS_Bar to present some necessary info to the code that needs it.

AsaNehm, 2024-09-10:
Files that have a hard-coded 35.42 in are (only in src)
- [ ] TMS_Bar.cpp
- [ ] TMS_Kalman.cpp
So in principle this should be easy to change. The bigger problem might be to change the call to where ever the width in xw or yw is referring to

AsaNehm, 2024-09-11:
The part in TMS_Bar.cpp is in addition inside a sanity check. So this could be taken out if we know that it works as it is supposed to

jdkio, 2024-09-13:
Yeah, TMS is far enough along that we don't need guardrails. We can let users use whatever values they want without any checks (we don't want constant warning messages in the logs)

We need to improve tms_geom so that it can return the right answer for any geometry questions. As a first step, we should at least replace all hard coded numbers to TMS geometry function calls. Then we can adjust the functions as needed. We need to think about how to best get the bar width out of the geometry manager

AsaNehm, 2024-09-23:
One additional thing is that the thickness of the scintillator bars is also in many places hardcoded to 10mm. With the new design having 17mm thickness this is going to be a problem as well

AsaNehm, 2024-09-23:
Took out the sanity check in TMS_Bar.cpp

jdkio, 2024-11-06:
@AsaNehm, looks like the sanity check in TMS_Bar is still there. Did you forget to make a PR?

AsaNehm, 2024-11-07:
Didn't forget it but wanted to leave it open for the Kalman filter stuff

closed 2025-02-04 by AsaNehm