DUNE/dune-tms #157: Kalman filter in main already?

jdkio, 2024-09-18:
When running main today I get a bunch of messages
```
Kalman filter start pos : 1840.12, -1982.16, 11368
Kalman filter end pos : 3282.12, -2024.93, 16873
Kalman filter start dir : 0.252851, -0.00802856, 0.967472
Kalman filter end dir : 0.0867626, -1.38628e-07, 0.996229
Kalman filter momentum: 5526.64 MeV
Kalman filter start pos : -961.194, -1818.2, 11368
Kalman filter end pos : -315.057, -2016.66, 14473
Kalman filter start dir : 0.196566, -0.0618855, 0.978536
Kalman filter end dir : 0.0865588, -0.103129, 0.990895
Kalman filter momentum: 6453.98 MeV
```
PR #122 isn't merged yet, and we don't want these numbers of messages anyway. So something's up

LiamOS, 2024-09-18:
Confirming there's an issue, I just went to view the Kalman PR and there's nothing to commit. One of the recent commits could have accidentally included this I guess, but unsure how we'd've missed that.

LiamOS, 2024-09-18:
I think @AsaNehm included the Kalman filter in #125

AsaNehm, 2024-09-18:
I tried to take the Kalman filter out as much as possible, giving you the option to take it in once you're ready to do so

closed 2025-02-04 by AsaNehm