DUNE/dune-tms #158: ConvertToTMSTree.exe crashes a lot

jdkio, 2024-09-18:
Running over 10 files of 
`/pnfs/dune/persistent/users/abooth/nd-production/MicroProdN1p2/output/run-spill-build/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu/EDEPSIM_SPILLS/0000000/0000100`
results in 3 crashes. At least it did on 9/12, with a slightly modified version of main. Modified in the sense that most changes made it into PR #152, and the rest were printing extra info. I'm getting a higher failure rate now using main

Try this and see what happens
```
for run in $(seq 100 110); do
echo "Running run ${run}:"
ConvertToTMSTree.exe /pnfs/dune/persistent/users/abooth/nd-production/MicroProdN1p2/output/run-spill-build/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu/EDEPSIM_SPILLS/0000000/0000100/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu.0000${run}.EDEPSIM_SPILLS.root &> regular_run_${run}.log
done 
```

jdkio, 2024-09-18:
This seems to be an AL9 issue. I'm getting a 100% crash rate for it. The SL7 version crashes sometimes (which we have to work on) but it's not 100%

jdkio, 2024-09-18:
Here's the gdb backtrace for one of the errors
```
#0  0x00007ffff4a8b94c in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x00007ffff4a3e646 in raise () from /lib64/libc.so.6
#2  0x00007ffff4a287f3 in abort () from /lib64/libc.so.6
#3  0x00007ffff4a29130 in __libc_message.cold () from /lib64/libc.so.6
#4  0x00007ffff4a959f7 in malloc_printerr () from /lib64/libc.so.6
#5  0x00007ffff4a9755c in _int_free () from /lib64/libc.so.6
#6  0x00007ffff4a99d35 in free () from /lib64/libc.so.6
#7  0x00007ffff7e67652 in TG4Event::~TG4Event (this=0x7ffffff9f4e0, __in_chrg=<optimized out>) at /tmp/losulliv/spack-stage/spack-stage-edep-sim-3.2.0-fnwu3lpodqtb6zh6eqko6uaxtg5fajo5/spack-src/io/TG4Event.cxx:7
#8  0x000000000040e6a5 in ConvertToTMSTree (filename=..., output_filename=...) at ConvertToTMSTree.cpp:143
#9  0x000000000040919d in main (argc=<optimized out>, argv=<optimized out>) at ConvertToTMSTree.cpp:292
```

jdkio, 2024-09-18:
For sl7, there's a wider variety of errors:
`ConvertToTMSTree.exe /pnfs/dune/persistent/users/abooth/nd-production/MicroProdN1p2/output/run-spill-build/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu/EDEPSIM_SPILLS/0000000/0000100/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu.0000123.EDEPSIM_SPILLS.root`
```
#6  0x000000000040f041 in std::_Destroy<TVector3> (__pointer=0x12d42920) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:98
#7  std::_Destroy_aux<false>::__destroy<TVector3*> (__last=<optimized out>, __first=0x12d42920) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:108
#8  std::_Destroy<TVector3*> (__last=<optimized out>, __first=<optimized out>) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:137
#9  std::_Destroy<TVector3*, TVector3> (__last=0x12d429e8, __first=<optimized out>) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:206
#10 std::vector<TVector3, std::allocator<TVector3> >::~vector (this=0x1387fa08, __in_chrg=<optimized out>) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_vector.h:677
#11 TMS_TrueParticle::~TMS_TrueParticle (this=0x1387f9d8, __in_chrg=<optimized out>) at ../src/TMS_TrueParticle.h:20
#12 std::_Destroy<TMS_TrueParticle> (__pointer=0x1387f9d8) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:98
#13 std::_Destroy_aux<false>::__destroy<TMS_TrueParticle*> (__last=<optimized out>, __first=0x1387f9d8) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:108
#14 std::_Destroy<TMS_TrueParticle*> (__last=<optimized out>, __first=<optimized out>) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:137
#15 std::_Destroy<TMS_TrueParticle*, TMS_TrueParticle> (__last=0x13887dd0, __first=<optimized out>) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_construct.h:206
#16 std::vector<TMS_TrueParticle, std::allocator<TMS_TrueParticle> >::~vector (this=0x7ffc3649e820, __in_chrg=<optimized out>) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_vector.h:677
#17 TMS_Event::~TMS_Event (this=0x7ffc3649e800, __in_chrg=<optimized out>) at ../src/TMS_Event.h:19
#18 0x000000000040acb3 in ConvertToTMSTree (filename=..., output_filename=...) at ConvertToTMSTree.cpp:216
#19 0x00000000004083e6 in main (argc=<optimized out>, argv=<optimized out>) at ConvertToTMSTree.cpp:292
```

`ConvertToTMSTree.exe /pnfs/dune/persistent/users/abooth/nd-production/MicroProdN1p2/output/run-spill-build/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu/EDEPSIM_SPILLS/0000000/0000100/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu.0000101.EDEPSIM_SPILLS.root`
```
*** Error in `ConvertToTMSTree.exe': free(): invalid next size (normal): 0x00000000112b7240 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81329)[0x7f78d2e97329]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN7TBufferD1Ev+0x2f)[0x7f78d
91b540f]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libRIO.so(_ZN11TBufferFileD0Ev+0x12)[0x7
f78d86da372]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN7TBranchD2Ev+0x268)[0x7f78
d66b63b8]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN14TBranchElementD0Ev+0x12)
[0x7f78d66c26e2]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN9TObjArray6DeleteEPKc+0x74
)[0x7f78d925da84]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN7TBranchD2Ev+0x164)[0x7f78
d66b62b4]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN14TBranchElementD0Ev+0x12)
[0x7f78d66c26e2]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN9TObjArray6DeleteEPKc+0x74
)[0x7f78d925da84]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN7TBranchD2Ev+0x164)[0x7f78
d66b62b4]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN14TBranchElementD0Ev+0x12)
[0x7f78d66c26e2]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN9TObjArray6DeleteEPKc+0x74
)[0x7f78d925da84]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN5TTreeD2Ev+0x1f1)[0x7f78d6
72d2f1]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN5TTreeD0Ev+0x12)[0x7f78d67
2d902]
ConvertToTMSTree.exe(_Z16ConvertToTMSTreeNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES4_+0x310e)[0x40b78e]
ConvertToTMSTree.exe(main+0xc6)[0x4083e6]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f78d2e38555]
ConvertToTMSTree.exe[0x40852d]
[... cut ...]
7f78d0bd2000-7f78d0bd3000 rw-p 00015000 00:9a 21846                      /usr/lib64/libz.so.1.2.7
7f78d0bd3000-7f78d0be1000 r-xp 00000000 00:9a 27344                      /usr/lib64/liblz4.so.1.8.3
7f78d0be1000-7f78d0de0000 ---p 0000e000 00:9a 27344                      /usr/lib64/liblz4.so.1.8.3
7f78d0de0000-7f78d0de1000 r--p 0000d000 00:9a 27344                      /usr/lib64/liblz4.so.1.8.3
7f78d0de1000-7f78d0de2000 rw-p 0000e000 00:9a 27344                      /usr/lib64/liblz4.so.1.8.3Aborted (core dumped)
```

jdkio, 2024-09-18:
I think part of these issues are when values are saved using a pass by reference. Like
https://github.com/DUNE/dune-tms/blob/main/src/TMS_TrueParticle.h#L69-L78

jdkio, 2024-09-18:
But modifying TrueParticle to not have those didn't fix it

LiamOS, 2024-09-18:
Have you had a chance to try an earlier commit? (e.g. 62232f6ab4b312d6ab62af5f676091cee41e5a84)
It's hard to believe we didn't break something in the last while if it stopped working on A9 and SL7 simultaneously.

I'll don't have that much time this week to focus on this, if we have to revert and split the desired commits into more manageable and reviewable parts we can do that. I can still update CAFMaker even if this side is still reviewing and merging.

jdkio, 2024-09-20:
Switched to [liam_lmao](https://github.com/DUNE/dune-tms/tree/liam_lmao). Writing notes here, feel free to head straight to my confirmation of the issue below

Starting with AL9, just to see. First cherry picked the setup fix.

On AL9, it finishes all the way with a bunch of "Not all hits in separated hit groups!" warnings.
```
Overlaying 15 events
Sliced event 1951 into 10 slices
Not all hits in separated hit groups!
Not all hits in separated hit groups!
```
Then it crashes
```
#5  0x00007f8ad077d060 in TBufferFile::WriteFastArray(int const*, long long) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#6  0x00007f8ad0a2c782 in int TStreamerInfo::WriteBufferAux<char**>(TBuffer&, char** const&, TStreamerInfo::TCompInfo* const*, int, int, int, int, int) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#7  0x00007f8ad085ae3d in TStreamerInfoActions::GenericWriteAction(TBuffer&, void*, TStreamerInfoActions::TConfiguration const*) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#8  0x00007f8ad078443d in TBufferFile::WriteClassBuffer(TClass const*, void*) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#9  0x00007f8acfc903ee in TBranch::Streamer(TBuffer&) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libTree.so.6.30
#10 0x00007f8ad078373b in TBufferFile::WriteObjectClass(void const*, TClass const*, bool) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#11 0x00007f8ad078b4f4 in TBufferIO::WriteObjectAny(void const*, TClass const*, bool) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#12 0x00007f8ad0d925ed in TObjArray::Streamer(TBuffer&) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libCore.so.6.30
#13 0x00007f8ad077e994 in TBufferFile::WriteFastArray(void*, TClass const*, long long, TMemberStreamer*) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#14 0x00007f8ad0a2ae12 in int TStreamerInfo::WriteBufferAux<char**>(TBuffer&, char** const&, TStreamerInfo::TCompInfo* const*, int, int, int, int, int) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#15 0x00007f8ad085ae3d in TStreamerInfoActions::GenericWriteAction(TBuffer&, void*, TStreamerInfoActions::TConfiguration const*) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#16 0x00007f8ad078443d in TBufferFile::WriteClassBuffer(TClass const*, void*) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#17 0x00007f8ad08344ac in TKey::TKey(TObject const*, char const*, int, TDirectory*) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#18 0x00007f8ad07e8ea5 in TFile::CreateKey(TDirectory*, TObject const*, char const*, int) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#19 0x00007f8ad07da2fe in TDirectoryFile::WriteTObject(TObject const*, char const*, char const*, int) () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libRIO.so.6.30
#20 0x00007f8ad0d0fb15 in TObject::Write(char const*, int, int) const () from /exp/dune/data/users/losulliv/september-spack/opt/spack/linux-almalinux9-x86_64_v3/gcc-11.4.1/root-6.30.08-mtmdkre34xhzbn3qya6uqf3vo6jcobjx/lib/root/libCore.so.6.30
#21 0x000000000040ed07 in TMS_TreeWriter::Write (this=0xbd4de0 <TMS_TreeWriter::GetWriter()::Instance>) at ../src/TMS_TreeWriter.h:37
#22 ConvertToTMSTree (filename=..., output_filename=...) at ConvertToTMSTree.cpp:243
#23 0x000000000040921d in main (argc=<optimized out>, argv=<optimized out>) at ConvertToTMSTree.cpp:272
```

Reading the output file, `Error in <TFile::ReadKeys>: reading illegal key, exiting after 2 keys`. It only has `Reco_Tree` and `Truth_Info` trees and they only have 1000 entries. This is from the auto save which is set at 1000. Branch_Lines (aka Line_Candidates) was never filled during the fill command. It's commented out `//Branch_Lines->Fill();` And there are fewer than 1000 spills so Truth_Spill  [... cut ...]
TMS_Events are deleted after we combine several of them, and it crashes so when we load the next event after merging. This isn't super consistent but it happens. Deleting all TLorentzVector info in them gets rid of the issue. Another fix that would allow us to keep the info is to add a destructor that clears those vectors. This fix worked for TMS_TrueParticle

I'm running on 20 files to test a larger sample (so far I'm 5/5). If that works, I'll try using a TMS_Event destructor instead so we can keep the neutrino info. Either way, I'll try to commit something tonight

jdkio, 2024-09-20:
Continuation of the notes. See the end section

I ran over 20 files. For non-zero exit codes, I moved the output to `/exp/dune/data/users/kleykamp/dune-tms/2024-09-19_test_liam_lmao`. Logs are stored in `logs`. This is running over the current microprod  `/pnfs/dune/persistent/users/abooth/nd-production/MicroProdN1p2/output/run-spill-build/MicroProdN1p2_NDLAr_1E18_RHC.spill.nu/EDEPSIM_SPILLS/0000000/0000000/`
```
Total files processed: 20
Nonzero exit codes: 3
Average time per command: 78 seconds
```
Much better but still not great. It crashes for runs 15 (exit code 134), 17 (139), and 18 (134).

Unfortunately, the errors for 15 and 18 are errors don't make it into the logs. Here's what I see on the screen. They are slightly different
15: ```*** Error in `ConvertToTMSTree.exe': double free or corruption (!prev): 0x000000000f820f40 ***```
exit code 134
18: ````*** Error in `ConvertToTMSTree.exe': corrupted size vs. prev_size: 0x000000000ff1b5e0 ***````

### 15 full error
Here's everything but the memory map. I added the first two lines from the log file
```
Processed 21/1969 (1.06653%)
timeout: the monitored command dumped core
*** Error in `ConvertToTMSTree.exe': double free or corruption (!prev): 0x000000000f820f40 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81329)[0x7fbfbe497329]
/cvmfs/dune.opensciencegrid.org/products/dune/./edepsim/v3_2_0/Linux64bit+3.10-2.17-e20-prof/lib/libedepsim_io.so(_ZN13TG4TrajectoryD1Ev+0xf8)[0x7fbfc7013b18]
/cvmfs/dune.opensciencegrid.org/products/dune/./edepsim/v3_2_0/Linux64bit+3.10-2.17-e20-prof/lib/libedepsim_io.so(_ZN4ROOT6Detail20TCollectionProxyInfo8PushbackISt6vectorI13TG4TrajectorySaIS4_EEE6resizeEPvm+0x60)[0x7fbfc7022490]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libRIO.so(_ZN19TGenCollectionProxy8AllocateEjb+0x13b)[0x7fbfc3d7cf3b]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN14TBranchElement20ReadLeavesCollectionER7TBuffer+0x1bb)[0x7fbfc1cbf84b]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN7TBranch8GetEntryExi+0xe1)[0x7fbfc1cb39e1]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN14TBranchElement8GetEntryExi+0x455)[0x7fbfc1cce8d5]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN14TBranchElement8GetEntryExi+0x116)[0x7fbfc1cce596]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN5TTree8GetEntryExi+0xbc)[0x7fbfc1d2a31c]
ConvertToTMSTree.exe(_Z16ConvertToTMSTreeNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES4_+0x7f7)[0x408c47]
ConvertToTMSTree.exe(main+0xc6)[0x4081b6]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7fbfbe438555]
ConvertToTMSTree.exe[0x4082fd]
```
```
addr2line -e ConvertToTMSTree.exe 0x408c47
/exp/dune/app/users/kleykamp/tms_liam_lmao/dune-tms/app/ConvertToTMSTree.cpp:114
```
This maps to `if (gRoo)` again. So my guess is this is another TLorentzVector3 type error that we haven't solved.

### 18 full error
```
Event loop took 79.406s for 2056 entries (0.0386216 s/entries)
TMS_TreeWriter wrote output to MicroProdN1p2_NDLAr_1E18_RHC.spill.nu.0000018.EDEPSIM_SPILLS_TMS_RecoCandidates_Hough_Cluster1.root
timeout: the monitored command dumped core
*** Error in `ConvertToTMSTree.exe': corrupted size vs. prev_size: 0x000000000ff1b5e0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x80c37)[0x7f74e7696c37]
/lib64/libc.so.6(+0x8120e)[0x7f74e769720e]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN9TObjArray6DeleteEPKc+0x74)[0x7f74eda5da84]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN7TBranchD2Ev+0xcf)[0x7f74eaeb621f]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN7TBranchD0Ev+0x12)[0x7f74eaeb6472]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN9TObjArray6DeleteEPKc+0x74)[0x7f74eda5da84]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN5TTreeD2Ev+0x1f1)[0x7f74eaf2d2f1]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libTree.so(_ZN5TTreeD0Ev+0x12)[0x7f74eaf2d902]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN5TList6DeleteEPKc+0xba8)[0x7f74eda57a48]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libCore.so(_ZN9THashList6DeleteEPKc+0x83)[0x7f74eda4d9f3]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libRIO.so(_ZN14TDirectoryFile5CloseEPKc+0xac)[0x7f74ecf3764c]
/cvmfs/larsoft.opensciencegrid.org/products/root/v6_22_08d/Linux64bit+3.10-2.17-e20-p392-prof/lib/libRIO.so(_ZN5TFile5CloseEPKc+0x15b)[0x7f74ecf5490b]
ConvertToTMSTree.exe(_Z16ConvertToTMSTreeNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES4_+0x36c0)[0x40bb10]
ConvertToTMSTree.exe(main+0xc6)[0x4081b6]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f74e7638555] 
ConvertToTMSTree.exe[0x4082fd]
```
So the crash happens after the files are completely written.
```
[... cut ...]
 values to be within the bounds but we need to think about what the right behavior is. I'm not sure what would happen if we chose not to write for values outside bounds, but then if all hits are outside bounds. In that case, would the accumulator be empty and then crash?

The TF1* is giving trouble in run 17 but I didn't look into it at all. Maybe sanitize might help there.

I've added `./RunOverSeveralFiles.sh` which I've been using to run over several files and gather crash rates. I'm currently running on the liam_lmao branch and outputing here: `/exp/dune/data/users/${USER}/dune-tms/2024-09-19_test_final`

jdkio, 2024-09-20:
`make sanitize` shows errors in `GetMuonTrueTrackLength`, but it seems to run when compiled with regular `make`. I'm at 22 files processed so far without crashes

jdkio, 2024-09-25:
This was fixed with PR #162. Closing issue

closed 2024-09-25 by jdkio