DUNE/dune-tms #170: PDR Readiness

jdkio, 2024-10-08:
Here's a list of things that need to be done according to [Hugh's production schedule](https://indico.fnal.gov/event/66222/contributions/300060/attachments/181792/249424/ProductionSchedule.pdf) from Sept 18th:
- [ ] Tagging TMS Reco
- [ ] Validating TMS Reco
- [ ] Tagging CAFMaker
- [ ] Validating CAFMaker
- [ ] TMS Stacking Plane
- [ ] TMS Module Orientation Plane
- [ ] Develop new TMS Geometry
- [ ] Validate TMS Geometry
- [ ] New sample production
- [ ] New sample validation

And then here's the [PDR requirements google doc](https://docs.google.com/document/d/1Aq7__N4_JFibsCWODixrQT0BWVe-AS5tDsfPWUl1aec/edit)
- [ ] Simulation tech note - Jeff, Alex
- [ ] Reco tech note - Asa, Liam, Jeff
- [ ] Event displays - 
- [ ] Acceptance - Kiyoung, Chris
- [ ] ND-LAr TMS track matching - Kate, Andy
- [ ] Muon momentum resolution - Jeff, Liam
- [ ] Muon sign selection - Xiaoyan, Liam
- [ ] Design optimization - TMS position optimization - Kiyoung, Chris
- [ ] Design optimization - TMS size optimization - Kiyoung, Chris
- [ ] Design optimization - Module orientation plan - Asa, Magnus

AsaNehm, 2024-11-06:
Created and validated the new geometry

AsaNehm, 2025-01-27:
Closing this as PDR is written by now

closed 2025-01-27 by AsaNehm