DUNE/dune-tms #201: Hybrid reco has multi-reco issues

jdkio, 2025-02-04:
This is post-x time slicing fix. Compare to stereo:
[image]
See [talk](https://indico.fnal.gov/event/67860/) and event displays

jdkio, 2025-02-04:
One of the requests was to remake this plot using only the ND physics sample. The idea being that many event displays showed a second track, likely a hadron. So this may not be an issue for hybrid events where only a muon makes it into TMS. I have a little work to do to add truth info to be able to do the ND physics sample correctly when it comes to reco efficiency type plots, though I can technically remake this plot correctly right now

AsaNehm, 2025-02-11:
I have an idea on what might cause the multi reconstruction of particles and would like to run this by you
What might happen is that when more than one track is found per orientation that all possibilities are tried out to match it with the other orientation(s). But after a match is made the 'simple track' doesn't get removed from the candidates list and more matches could be made if it satisfies the requirements.
This would explain why this behaviour appears for stereo and hybrid but way more often for the hybrid case as there are more options to make tracks with 3 orientations than just the two.

The fix would then be 1) to remove the already matched simple tracks from the candidates list or 2) to match other matching tracks into the same matched track
Option 1 is easier to implement but bears the risk of missing information
Option 2 is a bit harder to implement as it either requires a restructuring of the code or some semi-major extension to the code

AsaNehm, 2025-02-11:
Example of this in the output:    
    UTrack back: 0 | 46 | 6964.32 front: 30 | 26 | 6964.52
    VTrack back: 0 | 49 | 6957.54 front: 30 | 28 | 6967.51
    XTrack back: 0 | 169 | 6956.29 front: 30 | 153 | 6991.84
    Start: -1759.98 | 21.959 | 11250
    End: -2487.57 | -664.02 | 18535
    Added Direction: 53.926 | 171.495 | -1365
    UTrack back: 0 | 46 | 6964.32 front: 30 | 26 | 6964.52  // same UTrack as above
    VTrack back: 0 | 49 | 6957.54 front: 30 | 28 | 6967.51  // same VTrack as above
    XTrack back: 0 | 96 | 6924.86 front: 8 | 78 | 6923.32   // this is a different XTrack than above, but still gets matched
    Start: -1759.98 | 21.959 | 11250
    End: -2487.57 | -664.02 | 18535
    Added Direction: 53.926 | 171.495 | -1365
    UTrack back: 0 | 46 | 6964.32 front: 30 | 26 | 6964.52
    VTrack back: 0 | 40 | 6922.84 front: 7 | 39 | 6934.71
    XTrack back: 0 | 169 | 6956.29 front: 30 | 153 | 6991.84
    UTrack back: 0 | 46 | 6964.32 front: 30 | 26 | 6964.52
    VTrack back: 0 | 40 | 6922.84 front: 7 | 39 | 6934.71
    XTrack back: 0 | 96 | 6924.86 front: 8 | 78 | 6923.32
    UTrack back: 0 | 40 | 6923 front: 7 | 40 | 6929.43
    VTrack back: 0 | 49 | 6957.54 front: 30 | 28 | 6967.51
    XTrack back: 0 | 169 | 6956.29 front: 30 | 153 | 6991.84
    UTrack back: 0 | 40 | 6923 front: 7 | 40 | 6929.43
    VTrack back: 0 | 49 | 6957.54 front: 30 | 28 | 6967.51
    XTrack back: 0 | 96 | 6924.86 front: 8 | 78 | 6923.32
    UTrack back: 0 | 40 | 6923 front: 7 | 40 | 6929.43             // now a different UTrack than already matched twice
    VTrack back: 0 | 40 | 6922.84 front: 7 | 39 | 6934.71        // this is also a different VTrack than already matched
    XTrack back: 0 | 169 | 6956.29 front: 30 | 153 | 6991.84 // this is the same XTrack as in the first match made
    Start: -2047.58 | -1350 | 11250
    End: -2065.56 | -1692.99 | 12680
    Added Direction: 17.9753 | 342.99 | -1430                         // but gets matched again here
    UTrack back: 0 | 40 | 6923 front: 7 | 40 | 6929.43
    VTrack back: 0 | 40 | 6922.84 front: 7 | 39 | 6934.71
    XTrack back: 0 | 96 | 6924.86 front: 8 | 78 | 6923.32      // the same happens to the other XTrack here with the new U&VTracks
    Start: -2047.58 | -1334.23 | 11185
    End: -2056.57 | -1658.23 | 12680
    Added Direction: 17.9753 | 162 | -780

AsaNehm, 2025-02-11:
Option 3: introduce a figure of merit for which tracks get matched together

AsaNehm, 2025-02-18:
First approach (not optimal, but should solve the problem. Solution can and should be revisited in the future!!!):

- Sort 'simple' tracks by length / amount of hits in the track
- Start to merge the longest tracks first
- When match made, take 'simple' tracks in match out of the candidates list

AsaNehm, 2025-02-19:
I implemented this. @jdkio Could you check if this fix solved this issue for now? (branch 201-...)

closed 2025-03-20 by AsaNehm