DUNE/dune-tms #204: Hard-coded geometry 'changes'

AsaNehm, 2025-02-07:
List of variables and things that were manually changed, that could be taken from the geometry instead.

AsaNehm, 2025-02-07:
- [ ] app/CherryPickEvents.cpp: TMS_Thick_End -> Double_Thick_End. Instead refer to end of detector direct
- [ ] app/DrawEvents.cpp: Same as for CherryPickEvents.cpp
- [ ] config/TMS_Default_Config.toml: Fiducial 'volume' definition
- [ ] scripts/Reco/angle.cpp: positions of first scintillator plane, first thick scintillator plane and last scintillator plane (in z coordinates)
- [ ] scripts/Reco/draw_spill_3D_projections.py: already 'taken care of' in different issue
- [ ] scripts/Reco/enuqe.cpp: same as angle.cpp
- [ ] scripts/Reco/hadreco.cpp: same as angle.cpp
- [ ] scripts/Reco/make_hists.py: higher last z coordinate of 'detector' and first and last scintillator positions for tracks
- [ ] scripts/Reco/muonke.cpp: same as angle.cpp
- [ ] scripts/Reco/muonke_weights.cpp: same as angle.cpp
- [ ] scripts/Reco/muonke_weights_of.cpp: same as angle.cpp
- [ ] scripts/Reco/performance_reco.cpp: this is similar to draw_spill_3D_projections.py and highly dependend for plotting on the underlying geometry
- [ ] scripts/Validation/Plot_True_Track_Length.cpp: positions of first scintillator plane with new steel thickness and last scintillator plane overall
- [ ] scripts/Validation/Reco_Tracks.cpp: similar to make_hists.py
- [ ] scripts/Validation/Tracking_Validation.cpp: similar to angle.cpp

AsaNehm, 2025-02-07:
- [ ] src/TMS_Bar.cpp: TMS_Constant::TMS_Start_Exact->TMS_Start_Bars_Only [lines 135, 225]; hardcoded number to TMS_Const::TMS_Start_Bars_Only and use bar width [line 137]
- [ ] src/TMS_Constants.h: adding TMS_nDoublePlanes; changing numbers of planes; changing their positions changing number of modules, bars per module and overall plane number
- [ ] src/TMS_EventViewer.cpp: Thick_End -> Double_End
- [ ] src/TMS_Geom.h: changing spacing between planes? 11cm->13cm
- [ ] src/TMS_Reco.cpp: Thick_End->Double_End; adding in the double thick steel into the stereo check
- [ ] src/TMS_TreeWriter.cpp: Thick->Double_End

jdkio, 2025-02-26:
@AsaNehm, I was going to use something like 
```c++
geom->FindVolumeFast("volDetEnclosure");
```
but, as described in PR #208, that seems to be broken for the geometry we get from edep sim. We should replace all hardcoded geometry with calls to TMS_Geom, then have a flag somewhere to dictate which numberrs are returned. We might figure out a way to add a flag directly to the geometry, but for now we can adjust which numbers are used via a flag in [TMS_Default_Config.toml](https://github.com/DUNE/dune-tms/blob/main/config/TMS_Default_Config.toml)

# Possible ways of defining versions and stuff
```xml
<!-- Store version info as constants directly -->
<!-- Not sure if they get into the root geometry this way -->
<define>
    <constant name="GEOMETRY_VERSION" value="1.2.3"/>
    <constant name="TMS_GEOMETRY" value="Hybrid"/>
</define>
<!-- Or as an extra volume -->
<volume name="Detector_v1.2.3">
<volume name="GeometryType_v1.2.3">
<!-- This might work too? -->
<userinfo>
    <auxiliary auxtype="GeometryVersion" value="1.2.3"/>
</userinfo>
```
Apparently retrieved like this? I haven't tested it
```c++
TGeoManager *geo = gGeoManager;
TList *auxInfo = geo->GetTopVolume()->GetNodes()->First()->GetUserInfo();
```
Or here is the magic used to retrieve the Aux values used to store magnetic fields. We could store other values like geometry version info. It works in python after reading the gdmls directly. I haven't checked with the edep sim files directly. As I've said before, their geometry seems slightly corrupted so this may not work
```python
volumes = geom.GetListOfVolumes()
for vol in volumes:
    ue = vol.GetUserExtension()
    if ue != None and "tms" in vol.GetName().lower():
        user_obj = ue.GetUserObject()
        if isinstance(user_obj, cppyy.gbl.TMap):
            # Iterate over the keys in the TMap
            it = user_obj.MakeIterator()
            while it.Next():
                key = it.Key()
                value = it.Value()
                # Check if the key is 'BField'
                if key == "BField":
                    print(f"Found magnetic field: {value}")
                    # You can now parse the value if it's in a format like "(Bx, By, Bz)"
                    # Example: value = "(0 T, 1 T, 0 T)"
                    try:
                        bx, by, bz = map(float, str(value).strip('()').replace('T', '').split(","))
                        print(f"Parsed magnetic field for {vol.GetName()}:\tBx={bx}, By={by}, Bz={bz}")
                    except ValueError:
                        print("Failed to parse magnetic field value.")
```

kyjung123, 2025-02-27:
- [ ] src/TMS_Kalman.cpp, Where the B-filed is flipped. https://github.com/DUNE/dune-tms/blob/c10d7f51852f496e472c7198fd7904f9ed5bfef2/src/TMS_Kalman.cpp#L215

closed 2025-07-11 by AsaNehm