DUNE/dune-tms #207: Missing something in TrackMatching3D function

kyjung123, 2025-02-18:
I found that the RecoX (or RecoY) is not settled when we push back the start (or end) points in the hybrid case.
So there are quite many (0,0) points in the reco position. You can also see the 3D display.
[image]
For example, 
https://github.com/DUNE/dune-tms/blob/0f934d9681850ef4d44356edfcfd00f3fe8f482e/src/TMS_Reco.cpp#L898-L905
Here, we should add:
` UTracks.front().SetRecoX(0.5 * (VTracks.front().GetNotZ() + UTracks.front().GetNotZ()));                                                          
  UTracks.front().SetRecoY(CompareY(UTracks.front(), VTracks.front(), XTracks.front()));`
And the same edit for the following codes.

I can edit it, but before that, there is one thing I like to discuss.
In the TrackMatching function, basically, when the one track is push_back in the "aTrack" then we do --itU; like here,
https://github.com/DUNE/dune-tms/blob/0f934d9681850ef4d44356edfcfd00f3fe8f482e/src/TMS_Reco.cpp#L1403-L1405

However, here, https://github.com/DUNE/dune-tms/blob/0f934d9681850ef4d44356edfcfd00f3fe8f482e/src/TMS_Reco.cpp#L1417
there is only --itX without push_back. So I'm worried that there might be a case that didn't go through any conditions and just missed this X track.

AsaNehm, 2025-02-18:
Could you let me know which branch version is that you're using for this? I remember having fixed this already (the first thing you mentioned), at least in a couple of places
The second thing, I'll look into. In some cases it actually makes sense to do this. But will check to make sure

kyjung123, 2025-02-18:
I referred it from the main branch. Probably haven't applied yet?

AsaNehm, 2025-02-18:
Regarding the second thing mentioned in the issue: when looking closely through the code, one can see that the push_back call is included in all the different else (if) statements as one of the last lines without the decrease of the iterator afterwards. The one you see here in line 1417 is then the collective decrease, to avoid repeating this line over and over. Sorry for the confusion

kyjung123, 2025-02-18:
Oh... I see. So for itX you just decrease it collectively unlike U or V track. I see!
However, we are still gonna miss the Xtrack when itU==0 && itV==0. Right?
There are only conditions for (itV>0 && itU>0), (itV==0 && itU>0) and (itV>0 && itU==0).

AsaNehm, 2025-02-18:
I tried to fix this in branch 207-missing-something-in-trackmatching3d-function. Could you try if it works now?

kyjung123, 2025-02-18:
[image]
Looks good! Thanks, Asa.
But, What about the above statement? did you intend to skip the X track when itU==0&&itV==0?

AsaNehm, 2025-02-19:
I took care of this specific case in line 1149
`if ((UTracks[itU]).GetPlaneNumber() == (XTracks[itX]).GetPlaneNumber() || (itX > 0 && itU == 0 && itV == 0)) {`
The hits are then added in line 1181.
The only thing that would happen at the moment, that we don't really want is that the last U and V hit would in this constantly be added again, until no more X hits can be added

AsaNehm, 2025-02-19:
Fixed this using the `std::find` function: Looking through the hits in the track and see if the hit is already in there or not and only adding the U/V hit if they are not in

closed 2025-02-20 by AsaNehm