DUNE/dune-tms #217: Discuss Kalman filter

kyjung123, 2025-03-31:
I would like to discuss and confirm what I edited in the Kalman filter.
The result is from https://indico.fnal.gov/event/68849/contributions/312461/attachments/186448/256896/TMS_design032525.pdf

1. Wrong formula for the covariance matrix, in the main branch, the covariance matrix is just defined as https://github.com/DUNE/dune-tms/blob/1616bba7d9a59e31cb7c8a2c0e137b9285f9420d/src/TMS_Kalman.cpp#L381
However, when we update it to next state it should be like this `CovarianceMatrix = (IdentityMatrix-GainMatrix)*CovarianceMatrix;` 

[image] from arXiv:2404.08614

2. Too late Noise matrix, https://github.com/DUNE/dune-tms/blob/1616bba7d9a59e31cb7c8a2c0e137b9285f9420d/src/TMS_Kalman.h#L165
It is set as bar length, which is too large, and it make kalman filter be failed. see the presentation.

3. The Kalman filter missed the very first and last hit
When it makes the node the very last hit ( first hit in the tms) is not included,  https://github.com/DUNE/dune-tms/blob/1616bba7d9a59e31cb7c8a2c0e137b9285f9420d/src/TMS_Kalman.cpp#L51-L65
it needs to be changed like this:`if (abs(DeltaZ) > 1E-3||i==nCand-1)  `
                                                                                                                                   
Also, while kalman filtering, https://github.com/DUNE/dune-tms/blob/1616bba7d9a59e31cb7c8a2c0e137b9285f9420d/src/TMS_Kalman.cpp#L139-L146
It missed calling the State class for the very first node (last hit in the reco track). This led to losing the last hit in the Kalman position and the initial dxdz=1/999, which is initialized in the header file. 
Add this line before the loop, `KalmanNodes[0].SetRecoXY(KalmanNodes[0].PreviousState);` so it includes this hit in the reco position and call the TMS_KalmanState class.
Before fixing it 

[image]
Due to the first Kalman dxdx=1/999, it looks insane.

After fixing it 

[image]

There are more, such as smooth and charge, but once I clean up, I'll update it.

xiaoyan1997, 2025-03-31:
1, I think this is because Liam set the H matrix to be the identity matrix
2, I agree, some parameters need to be fine tuned
3, I think that is correct

kyjung123, 2025-04-10:
@xiaoyan1997 Yes, So it should be CovarianceMatrix = (IdentityMatrix-GainMatrix)*CovarianceMatrix with H matrix=1. or am I missing something?

LiamOS, 2025-04-14:
1) the H matrix is implicitly an identity matrix, since the quantities we measure directly correspond to those we fit.

2) Expanding the uncertainty to be the size of the bar was not something done for correctness, but out of being conservative. Kalman assumes that the uncertainty is Gaussian around the central values used, but with finite bars there's no "correct" value. From my understanding, if the errors are systematically overestimated the filter should still be able to converge, although slower, though if we re-run after an initial fit it should be able to converge on the same solution most of the time... I think.
As Xiaoyan said hasn't been studied in any real detail, but looking into the effects of this value are still useful, so thanks for the tests!

3) Good catch! 🥇 

4) The way it's done currently is just a hack to get most of it functioning, and was written when the reco information was a bit less fleshed out, so any improvements (and this does look like one) are welcome!