DUNE/dune-tms #222: Develop and main are diverging

jdkio, 2025-04-23:
Some PRs are being merged into main, and some into deveop. So they're diverging. We should maybe get rid of develop unless we're going to enforce no merging into main

LiamOS, 2025-04-29:
I'll have to look into branch protection to see if it's reasonably easily to make merging into develop the default.
When we had branch protection on I remember it being a bit flakey, but I also knew way less back then, so maybe it can just work.

jdkio, 2025-05-14:
I merged developed into main (thanks @LiamOS!). I'll run a test sample tomorrow and then we'll call that the next version for nersc

jdkio, 2025-05-14:
@LiamOS, @AsaNehm, I think there are issues with the merge
```bash
# Command
ConvertToTMSTree.exe /pnfs/dune/persistent/physicsgroups/dunendsim/abooth/nd-production/MicroProdN4p1/run-spill-build/MicroProdN4p1_NDComplex_FHC.spill.full/EDEPSIM_SPILLS/0002000/0002500/MicroProdN4p1_NDComplex_FHC.spill.full.0002500.EDEPSIM_SPILLS.root

# Seg fault:
#14 std::vector<int, std::allocator<int> >::vector (__x=..., this=0x15099548) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Li
nux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_vector.h:555
#15 TMS_TrueHit::TMS_TrueHit (other=..., this=0x150994e0) at TMS_TrueHit.h:23
#16 TMS_Hit::TMS_Hit (this=0x150994e0) at TMS_Hit.h:24
#17 std::_Construct<TMS_Hit, TMS_Hit const&> (__p=0x150994e0) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.
17/include/c++/9.3.0/bits/stl_construct.h:75
#18 std::__uninitialized_copy<false>::__uninit_copy<__gnu_cxx::__normal_iterator<TMS_Hit const*, std::vector<TMS_Hit, std::allocator<TMS_H
it> > >, TMS_Hit*> (__result=0x15098960, __last=..., __first=...) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.1
0-2.17/include/c++/9.3.0/bits/stl_uninitialized.h:83
#19 std::uninitialized_copy<__gnu_cxx::__normal_iterator<TMS_Hit const*, std::vector<TMS_Hit, std::allocator<TMS_Hit> > >, TMS_Hit*> (__re
sult=0x15098960, __last=..., __first=...) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0
/bits/stl_uninitialized.h:140
#20 std::__uninitialized_copy_a<__gnu_cxx::__normal_iterator<TMS_Hit const*, std::vector<TMS_Hit, std::allocator<TMS_Hit> > >, TMS_Hit*, T
MS_Hit> (__result=0x15098960, __last=..., __first=...) at /cvmfs/larsoft.opensciencegrid.org/products/gcc/v9_3_0/Linux64bit+3.10-2.17/incl
ude/c++/9.3.0/bits/stl_uninitialized.h:307
#21 std::vector<TMS_Hit, std::allocator<TMS_Hit> >::vector (this=0x7ffcca99e7e0, __x=...) at /cvmfs/larsoft.opensciencegrid.org/products/g
cc/v9_3_0/Linux64bit+3.10-2.17/include/c++/9.3.0/bits/stl_vector.h:555
#22 0x00007f4deb2edb28 in TMS_TrackFinder::TrackMatching3D (this=this
entry=0x2f41d40 <TMS_TrackFinder::GetFinder()::Instance>) at TMS_Reco.cpp:894
#23 0x00007f4deb2f4b7a in TMS_TrackFinder::FindTracks (this=this
entry=0x2f41d40 <TMS_TrackFinder::GetFinder()::Instance>, event=...) at TMS_Reco.cpp:627
#24 0x000000000040c1d1 in ConvertToTMSTree (filename=..., output_filename=...) at ../src/TMS_Reco.h:181
#25 0x0000000000409866 in main (argc=<optimized out>, argv=<optimized out>) at ConvertToTMSTree.cpp:271
```

jdkio, 2025-05-15:
@AsaNehm, @LiamOS We have a strict deadline of tomorrow morning so that the code can be run in time for the workshop. So I made the following tag:
`jdkio_2025-05-14_workshop_release_candidate_v1`
It's a tag of `jdkio_fix_vtracks_seg_fault`. But there seems to be a memory leak. I ran all 102 files and so far 43 have been held due to memory issues. I ran `2025-02-13_Kalman_simplest_B_field.tar.gz` by accident and I didn't get any memory issues. It's based on a version from about 3 months ago. It could from my change or not, I'm not sure

For a list of holds, see
https://fifemon.fnal.gov/monitor/d/000000146/why-are-my-jobs-held?from=now-24h&to=now&var-cluster=fifebatch&var-user=kleykamp&orgId=1
The job id is `76903468.0@jobsub01.fnal.gov`. I do see files in `/pnfs/dune/scratch/users/kleykamp/nd_production/2025-05-14_test_jdkio_2025-05-14_workshop_release_candidate_v1_test2`, but they seem corrupted. At least the ones I tried all have `probably not closed, trying to recover`. The script will try to copy any files that it can find regardless of whether the TMS code crashes. That's just in case it crashes right at the end

Running locally, the memory starts at around 10% and then it's already around 19% at 10% completion. That's about 2.2 GB on gpvm12. At 20% completion, it's still around 21% memory so it's expanding slowly. At 40% completion, it's at 24% memory which translates to about 2.6 GB. I ran with 4GB on the grid so I'm not sure why it got held but maybe it shows this isn't a perfect technique

Hopefully you can find the seg fault or Alex would need to run with an older version. Since `2025-02-13_Kalman_simplest_B_field` did work, that's the one I recommended