DUNE/dune-tms #226: Add nersc compatibility to CI

jdkio, 2025-05-15:
We're having issues on nersc specifically. We need to figure out exactly the environment at nersc and add that to the CI as close as possible.

LiamOS, 2025-05-15:
Assigning myself for now, is the plan to have the CI just emulate the build environment for now, or do we need to test it running?

jdkio, 2025-05-15:
When @alexbooth92 is not busy, he can send along the log to check if it was a nersc-specific issue or would be caught by the environment. [Here's](https://github.com/DUNE/ND_Production/blob/main/run-tms-reco/install_tms_reco.sh) the installation script for future reference. I don't see anything complex so maybe it was an issue of being on nersc specifically